OpenDriveLab/AgiBot-World #6: Extrinsic definition and depth of left and right cameras

zubair-irshad, 2025-01-02:
Hi authors, 

Great work and thanks for open sourcing the robotics dataset. This is going to be great for the robotics community. I have a few questions:

- Extrinsic definition: How are the extrinsic defined? i.e. with respect to what frame of reference? I see that for the following task and episode `327/648642/camera`, translation vectors of left and right hand are almost same. Is this expected? Also I am wondering how can the extrinsic parameter of the left and right hand camera with respect to a fixed base remain constant when the left and right hand are moving?

```
        "hand left translation_vector": [
            -0.0075380928276755745,
            -0.06436073473943844,
            0.04910818396009734
        ]
        
        "hand right translation_vector": [
            -0.00771181729353111,
            -0.06608699108764371,
            0.0496011481635894
        ]
```

- Thanks for releasing the depth maps for the head camera. I believe it would be quite useful for 3D manipulation research. Are there plans to release depth for left and right cameras as well?

- I tried visualizing the 3D pointcloud, extrinsic cameras and future states (i.e. `state/end/position` in 3D) and I couldn't get everything to align. Here I am transforming the point cloud by extrinsic camera of the head. Red denotes head camera, green and blue denote left and right hand cameras, rainbow dots denotes 16 future end positions of left and right hands. Do you know how well the quality of released extrinsic is? It could be that I am not visualizing them correctly here. I appreciate your response!

[image]

ModiShi, 2025-01-02:
Hello, 

thanks for your interest in our work

* **Q1:Extrinsic definition** 

  The extrinsic parameters of the waist camera are defined relative to the waist frame, and the extrinsic parameters of the head camera are defined relative to the head motor. **We will release our robot's URDF and a script to convert the camera extrinsics to the robot base frame and align them with each timestamp.** This will be available in mid-January.

* **Q2: Plans to Release Depth for Waist Cameras** 

  **Agibot World Alpha** does not include depth information for wrist cameras now, but we will consider to add wrist depth maps in future releases.

* **Q3: Quality of the Released Extrinsic: **

  We have verified the accuracy of the camera extrinsics and will provide a script for visualization along with the extrinsic conversion script.

Feel free to ask if you have any other questions or suggestions. We are continuously working on improving our dataset.

zubair-irshad, 2025-01-03:
Thanks for your prompt response to my questions! Looking forward to the visualization and extrinsic conversion scripts!

closed 2025-01-03 by ModiShi

zubair-irshad, 2025-02-07:
Hi Authors, Any new update on when is it going to be released? Thanks!

@ModiShi

ModiShi, 2025-02-26:
@zubair-irshad Sorry for the late reply, we’ve updated the extrinsic parameters on [Hugging Face](https://huggingface.co/datasets/agibot-world/AgiBotWorld-Alpha/tree/main/parameters). The parameters are now aligned to each image frame.

zubair-irshad, 2025-02-26:
Thanks a lot @ModiShi 

Are they supposed to be under parameters folder? When I download the file `648642-684757.tar` and untar it, it seems to give me videos and depth actually and not extrinsics, perhaps I am not looking in the right place?

Thanks!

ModiShi, 2025-03-03:
@zubair-irshad Yes, under the parameters folder

zubair-irshad, 2025-03-03:
@ModiShi I am looking here: https://huggingface.co/datasets/agibot-world/AgiBotWorld-Alpha/tree/main/parameters 

when I untar those files i see `depth` and `videos` subfolder and not extrinsics, is that supposed to be correct and should I be looking elsewhere? Thanks!

```
(base) ➜  648642 ls
depth  videos

```

ModiShi, 2025-03-03:
@zubair-irshad Seems there is no file `648642-684757.tar` at https://huggingface.co/datasets/agibot-world/AgiBotWorld-Alpha/tree/main/parameters, only 648533-664320.tar, 664322-676342.tar, 676346-687579.tar, 687581-704112.tar

[image]