OpenDriveLab/AgiBot-World #16: More clear documentation hugging face card

BeneChen, 2025-02-21:
Hi Team,

First, I want to express my immense appreciation for all of your hard work and contributions. However, I now have some concerns regarding the clarity of the tasks we’re working on. It would be incredibly helpful if we could maintain documentation that addresses the following points:
	1.	**Task Overview**: What does each task look like at first glance? (a short clip of the video is sufficient)
	2.	**Sensor Information**: What sensors are involved in each task? (I’ve noticed that some tasks don’t include tactile sensors.)
	3.	**Task Specifications**: Can we get a preview of the size and number of trajectories involved in each task?
The information provided before downloading the full dataset is currently too minimal. It isn’t easy to gauge what’s needed without downloading large files, which can be time-consuming and resource-intensive. A more detailed preview would help prevent unnecessary downloads.

Thanks again for your help, and I look forward to your feedback.

Best regards,
Chi Bene Chen

ModiShi, 2025-02-21:
Hi Chi Bene Chen,

Thank you for your thoughtful feedback—it’s truly helpful! We’re actively looking into ways to improve task visibility, including the possibility of a dedicated webpage with task previews, sensor details, and dataset specifications. This would allow users to better evaluate tasks before downloading. We’ll keep you updated as we make progress!

BeneChen, 2025-02-24:
Thanks for listening! A quick markdown file or CSV would be helpful for now, with the possibility of considering a webpage later. I also have a question about which tasks fall under the contact-rich manipulation category and include tactile sensors. It would be great if you could provide a list of Task IDs related to this.

ModiShi, 2025-02-25:
Thank you for your suggestions! Regarding tactile sensors, the data is still being processed and has not been uploaded yet. Tasks 666, 675, 676, 677, 694, 737, and 774 will include tactile information, and we’ll make it available soon on [AgiBotWorld-Beta](https://huggingface.co/datasets/agibot-world/AgiBotWorld-Beta).