OpenDriveLab/AgiBot-World #17: extrinsics of camera to robot base

wkMvg, 2025-02-25:
Hi Team,
Thank you very much for your open-source work. I would like to ask if it would be possible to provide the extrinsic parameters from each camera to the robot base to unify the coordinate systems of each camera? Looking forward to your reply. Thank you.

ModiShi, 2025-02-26:
We’ve recently updated the extrinsic parameters on [Hugging Face](https://huggingface.co/datasets/agibot-world/AgiBotWorld-Alpha/tree/main/parameters). The parameters are now aligned to each image frame.

ksrinivasan-tri, 2025-02-26:
Where can we also find the spec of the observations and actions for the ordering and meaning of the different feature dimensions?

ModiShi, 2025-02-27:
> Where can we also find the spec of the observations and actions for the ordering and meaning of the different feature dimensions?

On [Hugging Face](https://huggingface.co/datasets/agibot-world/AgiBotWorld-Alpha)

closed 2025-02-28 by wkMvg