OpenDriveLab/AgiBot-World #18: Data convert and visualize problem

wujiajin1, 2025-02-27:
I download sample dataset through huggingface website, After running:
```python scripts/convert_to_lerobot.py --src_path ./data/sample_dataset --task_id 390 --tgt_path ./data/sample_lerobot```
it prints:
```
(agibotworld) (base) lenovo@lenovo-GeekPro-17IRR:~/Downloads/AgiBot-World$ python scripts/convert_to_lerobot.py --src_path ./data/sample_dataset --task_id 390 --tgt_path ./data/sample_lerobot
Get Task Instruction <Scanning items at a supermarket checkout.A robot stands at the supermarket checkout counter, with three items and a barcode scanner on the counter. The items are randomly placed.>
Generating local dataset: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:11<00:00, 11.99s/it]
['Scanning items at a supermarket checkout.A robot stands at the supermarket checkout counter, with three items and a barcode scanner on the counter. The items are randomly placed.']
Generating dataset from raw dataset: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2101/2101 [00:00<00:00, 705486.57it/s]
Generating dataset from raw datasets:   0%|                                                                                                                                                                                | 0/1 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "/home/lenovo/Downloads/AgiBot-World/scripts/convert_to_lerobot.py", line 656, in <module>
    main(args.src_path, args.tgt_path, task_id, dataset_base, json_file, args.debug)
  File "/home/lenovo/Downloads/AgiBot-World/scripts/convert_to_lerobot.py", line 624, in main
    dataset.save_episode(task=episode_desc, videos=raw_dataset[1])
  File "/home/lenovo/Downloads/AgiBot-World/scripts/convert_to_lerobot.py", line 385, in save_episode
    raise ValueError()
ValueError
```
After that, run:
```python scripts/visualize_dataset.py --task-id 390 --dataset-path ./data/sample_lerobot/```
it prints:
```
(agibotworld) (base) lenovo@lenovo-GeekPro-17IRR:~/Downloads/AgiBot-World$ python scripts/visualize_dataset.py --task-id 390 --dataset-path ./data/sample_lerobot/
Traceback (most recent call last):
  File "/home/lenovo/Downloads/AgiBot-World/scripts/visualize_dataset.py", line 235, in <module>
    main()
  File "/home/lenovo/Downloads/AgiBot-World/scripts/visualize_dataset.py", line 230, in main
    dataset = LeRobotDataset(repo_id, root=root, local_files_only=True)
TypeError: LeRobotDataset.__init__() got an unexpected keyword argument 'local_files_only'
```
how to solve these problem?

wujiajin1, 2025-02-27:
ubuntu 22.04, All operations follow the instruction.

ModiShi, 2025-02-27:
Thanks for reporting this issue. Seems that the errors are related to the LeRobotDataset v2.1 updates, our code was developed and tested on LeRobotDataset v2.0 (commit [59e2757](https://github.com/huggingface/lerobot/commit/59e275743499c5811a9f651a8947e8f881c4058c)).You can try
`pip install git+https://github.com/huggingface/lerobot@59e275743499c5811a9f651a8947e8f881c4058c
`

zhaianbang, 2025-03-04:
task_index = self.meta.get_task_index(task)

        if not set(episode_buffer.keys()) == set(self.features):
            raise ValueError()

我也是在这里报错了，set(episode_buffer.keys()) 多了1个键 "task"。
set(episode_buffer.keys()) 和 set(self.features) 不相等，我们需要仔细分析这两个集合中的元素。根据您提供的输出信息，我们可以看到：

episode_buffer.keys():
包含的键有：
'observation.images.top_head'
'timestamp'
'observation.images.back_right_fisheye'
'episode_index'
'observation.images.head_center_fisheye'
'task_index'
'observation.images.hand_right'
'task'
'observation.images.cam_top_depth'
'observation.images.head_left_fisheye'
'observation.state'
'observation.images.back_left_fisheye'
'observation.images.head_right_fisheye'
'observation.images.hand_left'
'index'
'action'
'frame_index'
self.features:
包含的键有：
'observation.images.back_right_fisheye'
'task_index'
'observation.images.hand_right'
'observation.images.cam_top_depth'
'observation.state'
'observation.images.back_left_fisheye'
'observation.images.hand_left'
'observation.images.top_head'
'timestamp'
'episode_index'
'observation.images.head_center_fisheye'
'observation.images.head_left_fisheye'
'observation.images.head_right_fisheye'
'index'
'action'
'frame_index'
不相等的原因
额外的键:

episode_buffer 中的键 'task' 在 self.features 中没有对应的键。这意味着 episode_buffer 包含了一些在特征定义中没有列出的特征。

zhaianbang, 2025-03-04:
task_index = self.meta.get_task_index(task)

        if not set(episode_buffer.keys()) == set(self.features):
            raise ValueError()

我也是在这里报错了，set(episode_buffer.keys()) 多了1个键 "task"。
set(episode_buffer.keys()) 和 set(self.features) 不相等：

episode_buffer.keys():
包含的键有：
'observation.images.top_head'
'timestamp'
'observation.images.back_right_fisheye'
'episode_index'
'observation.images.head_center_fisheye'
'task_index'
'observation.images.hand_right'
'task'
'observation.images.cam_top_depth'
'observation.images.head_left_fisheye'
'observation.state'
'observation.images.back_left_fisheye'
'observation.images.head_right_fisheye'
'observation.images.hand_left'
'index'
'action'
'frame_index'
self.features:
包含的键有：
'observation.images.back_right_fisheye'
'task_index'
'observation.images.hand_right'
'observation.images.cam_top_depth'
'observation.state'
'observation.images.back_left_fisheye'
'observation.images.hand_left'
'observation.images.top_head'
'timestamp'
'episode_index'
'observation.images.head_center_fisheye'
'observation.images.head_left_fisheye'
'observation.images.head_right_fisheye'
'index'
'action'
'frame_index'
不相等的原因
额外的键:

episode_buffer 中的键 'task' 在 self.features 中没有对应的键。这意味着 episode_buffer 包含了一些在特征定义中没有列出的特征。

zhaianbang, 2025-03-04:
> Thanks for reporting this issue. Seems that the errors are related to the LeRobotDataset v2.1 updates, our code was developed and tested on LeRobotDataset v2.0 (commit [...]

这个方法可以，不再报下面的错了。
 if not set(episode_buffer.keys()) == set(self.features):
        raise ValueError()

```
git clone https://github.com/huggingface/lerobot.git
cd lebot
git checkout 59e275743499c5811a9f651a8947e8f881c4058c
pip install -e .
但是会出现下面的问题：
python scripts/convert_to_lerobot.py --src_path ./data/sample_dataset --task_id 390 --tgt_path ./data/sample_lerobot
Generating local dataset: 100%|██████████████████████████████████████████████████████████| 1/1 [00:09<00:00,  9.69s/it]
Generating dataset from raw dataset: 100%|█████████████████████████████████████| 2101/2101 [00:00<00:00, 837346.32it/s]
已杀死ting dataset from raw dataset:   0%|                                                    | 0/2101 [00:00<?, ?it/s]
崩掉，vscode的窗口也随着崩掉，笔记本电脑是微星强袭GE76-3080ti(16G显存)，2022年款。不知为何？

后面使用https://huggingface.co/datasets/agibot-world/AgiBotDigitalWorld这里面的教程，删掉只剩/AgiBotDigitalWorld/data/sample_dataset/meta_info/digitaltwin_5这一条（其他的digitaltwin_0 1 2这些都删掉，/AgiBotDigitalWorld/data/sample_dataset/observation也只剩digitaltwin_5），用来测试，可以顺利运行。

```

Cherryjingyao, 2025-03-05:
用这个方法解决以后出现的问题是，加载的dataset没有了’task'这个 feature

ModiShi, 2025-03-07:
> 用这个方法解决以后出现的问题是，加载的dataset没有了’task'这个 feature

There should be no "task" feature. 
If you want to get the task instruction when loading LeRobot Dataset:
```
dataset = LeRobotDataset(repo_id, root=root, local_files_only=True)
data = dataset[0]
task_instruction = dataset.meta.tasks[int(data["task_index"])]
```

ModiShi, 2025-03-07:
> > Thanks for reporting this issue. Seems that the errors are related to the LeRobotDataset v2.1 updates, our code was developed and tested on LeRobotDataset v2.0 (commit [59e2757](https://github.com/huggingface/lerobot/commit/59e275743499c5811a9f651a8947e8f881c4058c)).You can try `pip install git+https://github.com/huggingface/lerobot@59e275743499c5811a9f651a8947e8f881c4058c `
> 
> 这个方法可以，不再报下面的错了。 if not set(episode_buffer.keys()) == set(self.features): raise ValueError()
> 
> ```
> git clone https://github.com/huggingface/lerobot.git
> cd lerobot
> git checkout 59e275743499c5811a9f651a8947e8f881c4058c
> pip install -e .
> 但是会出现下面的问题：
> python scripts/convert_to_lerobot.py --src_path ./data/sample_dataset --task_id 390 --tgt_path ./data/sample_lerobot
> Generating local dataset: 100%|██████████████████████████████████████████████████████████| 1/1 [00:09<00:00,  9.69s/it]
> Generating dataset from raw dataset: 100%|█████████████████████████████████████| 2101/2101 [00:00<00:00, 837346.32it/s]
> 已杀死ting dataset from raw dataset:   0%|                                                    | 0/2101 [00:00<?, ?it/s]
> 崩掉，vscode的窗口也随着崩掉，笔记本电脑是微星强袭GE76-3080ti(16G显存)，2022年款。不知为何？
> 
> 后面使用https://huggingface.co/datasets/agibot-world/AgiBotDigitalWorld这里面的教程，删掉只剩/AgiBotDigitalWorld/data/
> ```

Could you add a breakpoint in the script to identify where the crash occurs

Cherryjingyao, 2025-03-10:
还有一个问题，数据标注里有lable_info有每个skill对应的起始和结束帧，有按照skill 拆分的脚本吗

ModiShi, 2025-03-12:
> 还有一个问题，数据标注里有lable_info有每个skill对应的起始和结束帧，有按照skill 拆分的脚本吗

We currently don't provide a script for this. However, you can easily filter out the specific skill segments by iterating through the `json` files. This process doesn't require loading any image data, so it's computationally efficient to implement.

Talnex, 2025-03-12:
Can I convert the original downloaded dataset in tar format to LeRobot format? It takes up a lot of space to unzip the tar file and convert it again.

ModiShi, 2025-03-13:
> Can I convert the original downloaded dataset in tar format to LeRobot format? It takes up a lot of space to unzip the tar file and convert it again.

No😞

Talnex, 2025-03-16:
If possible, can you provide the info.json file of each task? It takes more than 10 hours to calculate the mean, min, max and std of each task, and the resulting file is only 1kb.

Catchip, 2025-06-05:
"If anyone encounters the issue 'No matching distribution found for pyav>=12.0.5' while installing the '59e275743499c5811a9f651a8947e8f881c4058c' version of lerobot, try replacing 'pyav = ">=12.0.5"' with 'av = ">=12.0.5"' in the pyproject.toml file."

nqtabokado, 2025-06-13:
i installed lerobot in branch `59e275743499c5811a9f651a8947e8f881c4058c` and when I run convert to lerobot it gives me MemoryError
```
(agibotworld) D:\working\AgiBot-World>python scripts/convert_to_lerobot.py --src_path ./sample_dataset/ --task_id 390 --tgt_path ./a
Get Task Instruction <Scanning items at a supermarket checkout.A robot stands at the supermarket checkout counter, with three items and a barcode scanner on the counter. The items are randomly placed.>
Loading chunk 1/1: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:44<00:00, 44.79s/it]
Processing episodes in chunk:   0%|                                                                                                                | 0/1 [00:23<?, ?it/s]
Processing chunks:   0%|                                                                                                                           | 0/1 [01:08<?, ?it/s]
Traceback (most recent call last):
  File "D:\working\AgiBot-World\scripts\convert_to_lerobot.py", line 667, in <module>
    main(args.src_path, args.tgt_path, task_id, dataset_base, json_file, args.debug, args.chunk_size)
  File "D:\working\AgiBot-World\scripts\convert_to_lerobot.py", line 622, in main
    dataset.save_episode(task=episode_desc, videos=raw_dataset[1])
  File "D:\working\AgiBot-World\scripts\convert_to_lerobot.py", line 407, in save_episode
    self._save_episode_table(episode_buffer, episode_index)
  File "D:\working\lerobot\lerobot\common\datasets\lerobot_dataset.py", line 804, in _save_episode_table
    ep_dataset = datasets.Dataset.from_dict(episode_dict, features=self.hf_features, split="train")
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\datasets\arrow_dataset.py", line 953, in from_dict
    return cls(pa_table, info=info, split=split)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\datasets\arrow_dataset.py", line 682, in __init__
    self._fingerprint = generate_fingerprint(self)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\datasets\fingerprint.py", line 216, in generate_fingerprint
    hasher.update(state[key])
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\datasets\fingerprint.py", line 192, in update
    value_for_update = self.hash(value)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\datasets\fingerprint.py", line 188, in hash
    return cls.hash_bytes(dumps(value))
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\datasets\utils\_dill.py", line 109, in dumps
    dump(obj, file)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\datasets\utils\_dill.py", line 103, in dump
    Pickler(file, recurse=True).dump(obj)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\dill\_dill.py", line 420, in dump
    StockPickler.dump(self, obj)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\pickle.py", line 487, in dump
    self.save(obj)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\datasets\utils\_dill.py", line 70, in save
    dill.Pickler.save(self, obj, save_persistent_id=save_persistent_id)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\dill\_dill.py", line 414, in save
    StockPickler.save(self, obj, save_persistent_id)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\pickle.py", line 603, in save
    self.save_reduce(obj=obj, *rv)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\pickle.py", line 717, in save_reduce
    save(state)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\datasets\utils\_dill.py", line 70, in save
    dill.Pickler.save(self, obj, save_persistent_id=save_persistent_id)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\dill\_dill.py", line 414, in save
    StockPickler.save(self, obj, save_persistent_id)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\pickle.py", line 560, in save
    f(self, obj)  # Call unbound method with explicit self
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\dill\_dill.py", line 1217, in save_module_dict
    StockPickler.save_dict(pickler, obj)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\pickle.py", line 972, in save_dict
    self._batch_setitems(obj.items())
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\datasets\utils\_dill.py", line 83, in _batch_setitems
    dill.Pickler._batch_setitems(self, items)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\pickle.py", line 998, in _batch_setitems
    save(v)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\datasets\utils\_dill.py", line 70, in save
    dill.Pickler.save(self, obj, save_persistent_id=save_persistent_id)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\dill\_dill.py", line 414, in save
    StockPickler.save(self, obj, save_persistent_id)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\pickle.py", line 603, in save
    self.save_reduce(obj=obj, *rv)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\pickle.py", line 692, in save_reduce
    save(args)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\datasets\utils\_dill.py", line 70, in save
    dill.Pickler.save(self, obj, save_persistent_id=save_persistent_id)
[... cut ...]
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\site-packages\dill\_dill.py", line 414, in save
    StockPickler.save(self, obj, save_persistent_id)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\pickle.py", line 560, in save
    f(self, obj)  # Call unbound method with explicit self
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\pickle.py", line 803, in save_bytes
    self._write_large_bytes(BINBYTES + pack("<I", n), obj)
  File "C:\Users\minht\miniconda3\envs\agibotworld\lib\pickle.py", line 260, in write_large_bytes
    write(payload)
MemoryError
```

Catchip, 2025-06-13:
@nqtabokado This phenomenon is most likely due to insufficient memory. You can try modifying convert_to_lerobot.py by removing data sections you might not need, such as depth maps or most of the fisheye camera videos.

iwinnter, 2025-07-18:
> If anyone encounters the issue `No matching distribution found for pyav>=12.0.5` while installing the '59e275743499c5811a9f651a8947e8f881c4058c' version of lerobot.如果有人遇到问题，第一名在安装“59e275743499c5811a9f651a8947e8f881c4058c”版本的Android机器人时，
> 
> try replacing `pyav = ">=12.0.5"` with `av = ">=12.0.5"` in the pyproject.toml file.尝试在pyproject.toml文件中用`pyav = ">=12.0.5"`替换`av = ">=12.0.5"`。

I've also run into the same problem. Could you tell me how you solved it?

> If anyone encounters the issue `No matching distribution found for pyav>=12.0.5` while installing the '59e275743499c5811a9f651a8947e8f881c4058c' version of lerobot.
> 
> try replacing `pyav = ">=12.0.5"` with `av = ">=12.0.5"` in the pyproject.toml file.

I've also run into the same problem. Could you tell me how you solved it?

Catchip, 2025-07-18:
@iwinnter try replacing pyav = ">=12.0.5" with av = ">=12.0.5" in the pyproject.toml file.