OpenDriveLab/AgiBot-World #20: extrinsics of camera iin AgiBot-DigitalWorld

DekuLiuTesla, 2025-03-01:
Hi Team, thank you very much for this excellent work! 

I'm trying the sample dataset of AgiBot-DigitalWorld. And I would like to know if there is any way to get the per-frame extrincs of each camera. Take the robot `head` as an example, I noticed that only a single extrinsic is provided in `camera_parameter.json`. And the 2DoF provided in `/state/joint/position` of `proprio_joints.h5` is apparently not enough to describe the pose. 

Therefore, I'm wondering if there's any way to get per-frame parameter as mentioned in https://github.com/OpenDriveLab/AgiBot-World/issues/17? Looking forward to your reply. Thank you.

ModiShi, 2025-03-07:
1. Working on it! We will update the extrinsic parameters soon.
2. The shape of `/state/joint/position` in `proprio_joints.h5` should be [N, 34], which episode did you try?

DekuLiuTesla, 2025-03-07:
@ModiShi Thanks for your reply! 1. That's great! I'm looking forward to it! 2. Yes, the shape is [N, 34]. Here, I mean there are 2 out of 34 dimensions used to describe state of `head` so I think it can be converted to extrinsics. But my issue has already been resolved!

closed 2025-03-07 by DekuLiuTesla