OpenDriveLab/AgiBot-World #26: Loading data speed

ksrinivasan-tri, 2025-03-06:
I find that loading the AgiBotDataset as shown in the ipython notebook

```
# =============================================
# 2. Dataset Setup
# =============================================
observation_idx = np.array([-1, 0])
action_idx = np.arange(-1, 15)
repo_id = f"agibotworld/task_{TASK_ID}"

delta_timestamps = {
    "observation.images.top_head": (observation_idx / FPS).tolist(),
    "observation.state": (observation_idx / FPS).tolist(),
    "action": (action_idx / FPS).tolist(),
}

dataset = LeRobotDataset(
    repo_id=repo_id,
    root=f"{dataset_path}/{repo_id}",
    delta_timestamps=delta_timestamps,
    local_files_only=True
)

dataloader = torch.utils.data.DataLoader(
    dataset,
    num_workers=0,
    batch_size=64,
    shuffle=True,
    pin_memory=(device.type == "cuda"),
    drop_last=True,
)
```
can sometimes take on the order of 50mins to 6 hours depending on the size of the dataset. Are there faster approaches (for instance using streaming) that you can provide any examples for?

ksrinivasan-tri, 2025-03-06:
I have tried loading the hf_dataset by overwriting the load_dataset argument `streaming=true` as shown [here](https://github.com/ksrinivasan-tri/AgiBot-World/blob/resume-convert/scripts/convert_to_lerobot.py#L369), however, this causes a later error inside huggingface datasets due to instantiating a IterableDataset instance that does not implement `__getitem__`. 

Traceback:
```
---------------------------------------------------------------------------
NotImplementedError                       Traceback (most recent call last)
/tmp/ipykernel_285793/4185017502.py in <module>
      5 meta = LeRobotDatasetMetadata(repo_id, dataset_root, True)
      6 episode_data_index = get_episode_data_index(meta.episodes, None)
----> 7 check_timestamps_sync(dataset, episode_data_index, 30, 1e-4)

~/.local/lib/python3.10/site-packages/lerobot/common/datasets/utils.py in check_timestamps_sync(hf_dataset, episode_data_index, fps, tolerance_s, raise_value_error)
    364     account for possible numerical error.
    365     """
--> 366     timestamps = torch.stack(hf_dataset["timestamp"])
    367     diffs = torch.diff(timestamps)
    368     within_tolerance = torch.abs(diffs - 1 / fps) <= tolerance_s

~/.local/lib/python3.10/site-packages/torch/utils/data/dataset.py in __getitem__(self, index)
     61 
     62     def __getitem__(self, index) -> _T_co:
---> 63         raise NotImplementedError("Subclasses of Dataset should implement __getitem__.")
     64 
     65     # def __getitems__(self, indices: List) -> List[_T_co]:

NotImplementedError: Subclasses of Dataset should implement __getitem__.
```

ModiShi, 2025-03-07:
@ksrinivasan-tri  The slow loading may be due to depth images being loaded individually. You can try converting depth images to .mp4 format and updating the corresponding feature in LeRobot to use video instead of image for faster loading.