OpenDriveLab/AgiBot-World #28: Error in camera extrinsic

notFoundThisPerson, 2025-03-12:
Thanks for your excellent work on the dataset. However, I noticed a consistent error in the camera extrinsics. When I project the point cloud from the head camera to the right-hand camera, the image appears to shift to the up right direction. Could there be a difference in the optical frame poses between the RGB and depth channels, or are there other pose translation issues?

Here is the code I used:
```python
import imageio
import numpy as np
import json
from PIL import Image

video_head = imageio.get_reader('observations/327/648642/videos/head_color.mp4')
head_rgb = video_head.get_data(0)
video_head.close()

video_hand = imageio.get_reader('observations/327/648642/videos/hand_right_color.mp4')
hand_rgb = video_hand.get_data(0)
video_hand.close()

head_depth = np.array(Image.open('observations/327/648642/depth/head_depth_000000.png')) / 1000.

with open('parameters/327/648642/parameters/camera/head_extrinsic_params_aligned.json', 'r') as f:
    head_extrinsic = json.load(f)[0]['extrinsic']
    head_cam_pose = np.eye(4)
    head_cam_pose[:3, 3] = np.array(head_extrinsic['translation_vector'])
    head_cam_pose[:3, :3] = np.array(head_extrinsic['rotation_matrix'])
with open('parameters/327/648642/parameters/camera/hand_right_extrinsic_params_aligned.json', 'r') as f:
    hand_extrinsic = json.load(f)[0]['extrinsic']
    hand_cam_pose = np.eye(4)
    hand_cam_pose[:3, 3] = np.array(hand_extrinsic['translation_vector'])
    hand_cam_pose[:3, :3] = np.array(hand_extrinsic['rotation_matrix'])
with open('parameters/327/648642/parameters/camera/head_intrinsic_params.json', 'r') as f:
    head_intrinsic = json.load(f)['intrinsic']
with open('parameters/327/648642/parameters/camera/hand_right_intrinsic_params.json', 'r') as f:
    hand_intrinsic = json.load(f)['intrinsic']
    
# build point cloud
h, w = head_depth.shape
u, v = np.meshgrid(np.arange(w), np.arange(h))
x = (u - head_intrinsic['ppx']) / head_intrinsic['fx'] * head_depth
y = (v - head_intrinsic['ppy']) / head_intrinsic['fy'] * head_depth
cloud = np.stack([x, y, head_depth], axis=2).reshape(-1, 3)
color = head_rgb.reshape(-1, 3)

# transpose point cloud from head camera coord to hand camera coord
cam_pose_trans = np.matmul(np.linalg.inv(hand_cam_pose), head_cam_pose)
cloud_hand = np.matmul(cloud, cam_pose_trans[:3, :3].T) + cam_pose_trans[None, :3, 3]

# project the point cloud to hand camera
flag = cloud_hand[:, 2] > 0
cloud_hand = cloud_hand[flag]
color = color[flag]
u_hand = np.round(cloud_hand[:, 0] / cloud_hand[:, 2] * hand_intrinsic['fx'] + hand_intrinsic['ppx']).astype(np.int32)
v_hand = np.round(cloud_hand[:, 1] / cloud_hand[:, 2] * hand_intrinsic['fy'] + hand_intrinsic['ppy']).astype(np.int32)

# get the projected image
img_proj = np.zeros_like(hand_rgb)
h, w, _ = img_proj.shape
flag = np.all(np.stack([u_hand >= 0, u_hand < w, v_hand >= 0, v_hand < h], axis=0), axis=0)
u_hand, v_hand, color = u_hand[flag], v_hand[flag], color[flag]
dist = cloud_hand[flag, 2]
idx = np.argsort(dist)
for i in idx:
    img_proj[v_hand[i], u_hand[i]] = color[i]
Image.fromarray(img_proj).save('proj.png')
Image.fromarray(hand_rgb).save('hand.png')
Image.fromarray(head_rgb).save('head.png')
```

[image]
[image]

MBronars, 2025-03-16:
I have been having similar issues trying to project points onto the camera frame based on proprioceptive states and camera params.  Were you able to fix this at all?

ModiShi, 2025-03-27:
We've identified some issues in the extrinsic parameters, we plan to fix them and upload corrected data soon, expected to be completed within 1-2 weeks.

LibertFan, 2025-04-12:
> We've identified some issues in the extrinsic parameters, we plan to fix them and upload corrected data soon, expected to be completed within 1-2 weeks.

Any Update?

jiahao-shao1, 2025-04-20:
Authors, this is excellent work!

I've attempted to reproduce the results using the provided script provided by @notFoundThisPerson and the updated camera parameters available at [https://huggingface.co/datasets/agibot-world/AgiBotWorld-Alpha/blob/main/parameters/648533-663816.tar]. However, I seem to be encountering an issue where the output is still not as expected.

[image]

[image]

vectorNN, 2025-06-23:
After updating the parameters, the intrinsic parameters were also updated. The image resolution is 640×480, but the intrinsic parameters are as follows:
{
"intrinsic": {
"fx": 644.523681640625,
"fy": 643.7030029296875,
"ppx": 647.5728759765625,
"ppy": 365.7279052734375,
"distortion_model": "plumb bob"
}
}
Why are the optical center coordinates (cx, cy) in the intrinsic parameters so large? Did we upload the parameters incorrectly?