OpenDriveLab/AgiBot-World #34: Example to use camera intrinsic and extrinsic parameters

jianzhang96, 2025-03-17:
Hi Team, 
Thanks for your excellent work.

I am trying to visualize the “state/end/position” on the image from the head camera using intrinsic and extrinsic parameters, but I failed.
Could you please share a script or code that shows how to use intrinsic and extrinsic parameters to project a 3D position onto a 2D image or to reconstruct a point cloud?

wkMvg, 2025-03-20:
> [@jianzhang96](https://github.com/jianzhang96) Hi, where are your camera paramters from? I downloaded data from opendatalab, and cannot find any related files. Could you show your file structure? or the camera parameters example?

在hf上 agibot的数据集里有 opendatalab上没有更新

LibertFan, 2025-04-12:
> Hi Team, Thanks for your excellent work.
>
> I am trying to visualize the “state/end/position” on the image from the head camera using intrinsic and extrinsic parameters, but I failed. Could you please [...]

请问你现在可视化成功了吗？