OpenDriveLab/AgiBot-World #35: Action and state meaning

lzl2040, 2025-03-18:
The dimension of the ```action``` and ```observation.state``` is 22 and 20, respectively. Can you tell me the meaning of each dimension? Thank you!

ModiShi, 2025-03-19:
In the process of [converting the dataset to lerobot format](https://github.com/OpenDriveLab/AgiBot-World/blob/47e70b59c3f13a53449efe27e25ac6a764409185/scripts/convert_to_lerobot.py#L498-L517), we preserved the following dimensions:
- 14(2*7) DOF for dual arms
- 2 DOF for gripper states (opening/closing)
- 2 DOF for head movement
- 2 DOF for waist movement

For the action space, we added 2 additional DOF for the robot base movement, bringing the total to 22 dimensions.

For more detailed information about the meaning of each specific dimension in the original data, please refer to [Hugging Face](https://huggingface.co/datasets/agibot-world/AgiBotWorld-Alpha).

closed 2025-03-19 by ModiShi

lzl2040, 2025-04-07:
Thank you for your reply? I want to transform the joint positions into the ```x, y, z, quat (4D)```. Do you know how to solve this?

LibertFan, 2025-04-12:
> Thank you for your reply. I want to transform the joint positions into the `x, y, z, quat (4D)`. Do you know how to solve this?

Have you solved this? I have the same needs.