OpenDriveLab/AgiBot-World #38: LeRobotDataset v2.0 (commit [59e2757])

Mcccccc1024, 2025-03-23:
> Thanks for reporting this issue. Seems that the errors are related to the LeRobotDataset v2.1 updates, our code was developed and tested on LeRobotDataset v2.0 (commit [59e2757](https://github.com/huggingface/lerobot/commit/59e275743499c5811a9f651a8947e8f881c4058c)).You can try
> `pip install git+https://github.com/huggingface/lerobot@59e275743499c5811a9f651a8947e8f881c4058c
> ` 

 _Originally posted by @ModiShi in [#18](https://github.com/OpenDriveLab/AgiBot-World/issues/18#issuecomment-2687418549)_

Hello, this version of LeRobot is no longer available on Hugging Face. Could you please provide this version? Thank you.

ModiShi, 2025-03-27:
If you can't intall lerobot through `pip install git+https://github.com/huggingface/lerobot@59e275743499c5811a9f651a8947e8f881c4058c`, you can try
```
git clone https://github.com/huggingface/lerobot.git
cd lerobot
git checkout 59e275743499c5811a9f651a8947e8f881c4058c
pip install -e .
```

fresh-men, 2025-04-27:
> If you can't intall lerobot through `pip install git+https://github.com/huggingface/lerobot@59e275743499c5811a9f651a8947e8f881c4058c`, you can try
> 
> ```
> git clone https://github.com/huggingface/lerobot.git
> cd lerobot
> git checkout 59e275743499c5811a9f651a8947e8f881c4058c
> pip install -e .
> ```

我按照这个方式安装后，仍然有问题：FileNotFoundError: [Errno 2] No such file or directory: 'ffprobe'

huxuan, 2025-04-27:
> 我按照这个方式安装后，仍然有问题：FileNotFoundError: [Errno 2] No such file or directory: 'ffprobe'

`ffprobe` is a part of `ffmpeg`，please refer to the official documentation for installation: https://ffmpeg.org/download.html

yuhongwei22, 2025-05-06:
> 我按照这个方式安装后，仍然有问题：FileNotFoundError: [Errno 2] No such file or directory: 'ffprobe'

you can try "sudo apt-get install  ffmpeg". I solve the problem by this command.

ksrinivasan-tri, 2025-05-08:
This will no longer work due to the renaming of the package pyav to av, the older version of lerobot was depending on a version of pyav due to the use of torchvision==0.17. However, now that the package has been renamed on pypi it can no longer be pip installed, and therefore the command:
`pip install git+https://github.com/huggingface/lerobot@59e275743499c5811a9f651a8947e8f881c4058c` no longer works.
@yuhongwei22 @ModiShi Can you both please look into fixing the conversion script to update to the latest lerobot version?

The error I get when using the latest lerobot commit version is:
```
Loading chunk 1/8: 100%|______________________________________________________________________________________________________________________________________________________________________________________________________________________| 10/10 [00:35<00:00,  3.60s/it]
Processing episodes in chunk:   0%|                                                                                                                                                                                                                    | 0/10 [00:00<?, ?it/s]
Processing chunks:   0%|                                                                                                                                                                                                                                | 0/8 [00:36<?, ?it/s]
Traceback (most recent call last):                                                                                                                                                                                                                                            
  File ".../lerobot_venv/convert_to_lerobot.py", line 1085, in <module>
    main(
  File ".../lerobot_venv/convert_to_lerobot.py", line 1036, in main
    dataset.save_episode(task=task_name, videos=raw_dataset[1])
  File ".../lerobot_venv/convert_to_lerobot.py", line 595, in save_episode
    raise ValueError()
ValueError
```

ModiShi, 2025-05-11:
@ksrinivasan-tri try [this](https://github.com/Tavish9/any4lerobot)