OpenDriveLab/AgiBot-World #45: Camera external parameters and robotic arm pose

Yingminj, 2025-04-24:
I would like to ask:
1. Why is there a difference between the head camera's extrinsic parameters (head_extrinsic_params.json) and the frame-by-frame extrinsic parameter values ​​(head_extrinsic_params_aligned.json)? 
2. 2. Is the coordinate system of the head camera's extrinsic calibration the same as the coordinate system of the left and right hand poses? 
3. Because I want to get the relationship between the end pose and the head camera image coordinates for calculation. 

Thank you!!

ChengenXie, 2025-04-28:
Hi, thanks for your interest in our work! Just sending over my response to your question:

The extrinsic parameters in head_extrinsic_params.json are the default setting for the head perspective. However, the head perspective will vary during data collection, so please use the frame-by-frame extrinsic parameters in 'head_extrinsic_params_aligned.json'.
Yes, the head camera's extrinsic and hand poses are defined in the same coordinate system. Note ：This extrinsic is not the transformation matrix from the ego coordinate system to the camera coordinate system. So you should use the following formula camera_pos = rotation_matrix.T @ (current_point_copy -translation_vector.reshape).
Hope that helps! Let me know if you have any more questions.

yanziwei, 2025-11-03:
How is head_extrinsic_params_aligned.json calculated?