OpenDriveLab/AgiBot-World #63: Misalignment in the usage of wogripper parameter between fine-tuning and inference

LostS0u1, 2025-07-03:
It seems that there is a misalignment in the usage of the wogripper parameter between the fine-tuning and inference phases. We would like to clarify how this parameter should be properly used across different scenarios.

Thanks.

LostS0u1, 2025-07-03:
It seems that in infer.py of genie_sim it is by default wogripper=Ture.  While in finetune.py, the dafault value is False。

closed 2025-07-04 by LostS0u1

linxcan, 2025-08-09:
Hello, if I use wogripper=Ture, does that mean I won't train the gripper?

ReubenLimMonash, 2025-08-20:
Hi, I would also like to ask regarding this. If we set --wogripper when running the finetuning script, does that mean the gripper isn't trained? How would we then control the gripper during inferencing?