OpenDriveLab/AgiBot-World #76: "key_frame" is not found in task json.

Tonniia, 2025-07-10:
Hi, here is the task_327.json file I downloaded from AgiBotWorld-Beta:

[image]

It appears that there is no item named "key_frame" in the file.
Additionally, when comparing it with the example provided on Hugging Face:

[image]
I noticed some differences. Moreover, I couldn't find the entry "episode_id": 649078 in my task_327.json.
Has there been a recent update to AgiBotWorld-Beta?

ModiShi, 2025-07-10:
Thanks for catching that! We've removed the "key_frame" field since the action_config already includes the start and end frames. Please refer to the current dataset version as the reference.

Tonniia, 2025-07-11:
Thanks😊 Hope for release of failure case as negative sample during training. For example, two cases with same trajector, but one case grisper not closed enough, thus the object cannot be successfully picker up.

closed 2025-07-11 by Tonniia