OpenDriveLab/AgiBot-World #83: Real world robot do not match simulation robot

junbangliang, 2025-07-21:
Hi,

I noticed that the simulation robot end effector dimension is very different from the real robot in distance from flange to the gripper tip. Does that mean the simulation is no good for validating policies train with the real-world dataset?

[image]

[image]

Thanks!

ModiShi, 2025-07-23:
Our challenge data and benchmark only use the white gripper and don't support switching to the black gripper. The black gripper was only used in some previous open-source datasets.

junbangliang, 2025-07-23:
I see, only the iros challenge uses the white gripper and the alpha and beta dataset and rest of the benchmark uses the black gripper. Thanks!

closed 2025-07-23 by junbangliang