OpenDriveLab/AgiBot-World #92: Problem of converting 3D coordinates to pixel coordinates and the issue with camera intrinsic and extrinsic parameters.

mingyi-lee, 2025-08-07:
Thank you for your kind words! I'm actually encountering some issues while working with the sample-dataset. Specifically, I'm trying to visualize the end-position onto pixel coordinates. I attempted to use the camera's intrinsic and extrinsic parameters for this purpose, but the visualization results were not correct. Later, I came across a formula on Hugging Face: `camera_pos = rotation_matrix.T @ (current_point_copy - translation_vector.reshape)`, but the visualization still didn't turn out right. I was wondering if there might be a solution to this problem. Thank you for your time and help!

leeibo, 2025-08-20:
您好，请问问题解决了吗

venchi012, 2025-09-15:
同样的问题