OpenDriveLab/AgiBot-World #112: 可以直接跳过微调启动模型，查看效果吗？希望可以更详细的步骤介绍

bigsmartbig, 2025-09-28:
[image]

bigsmartbig, 2025-09-28:
data_stats_path是指什么的路径？传入前面下载的LIBERO数据集路径可以吗

bigsmartbig, 2025-09-29:
Could you explain what each parameter in the payload? Or where can I learn about them?

ModiShi, 2025-09-29:
By default, the model takes three images, one instruction, and robot state (proprioception) as input to output actions. Different LeRobot datasets use different image key names (e.g., "image", "head_color"), which need to be specified in the [config](https://github.com/OpenDriveLab/AgiBot-World/blob/main/go1/configs/go1_sft_agilex.py#L63-L71). You can also use fewer than 3 images, as shown in our [Libero example](https://github.com/OpenDriveLab/AgiBot-World/blob/main/go1/configs/go1_sft_libero.py#L63-L71). The `ctrl_freqs` parameter sets the robot control frequency - use the default value of 30 if unsure.